clockworkpi/uConsole #2: R-01 cursor missing in multi-user mode.

markatlnk, 2023-07-18:
Going for as much performance as I can get, decided to run without the GUI. Text is really small, but the main issue is there isn't a cursor. Makes it hard to do many things.

It is hard to get this into text only mode, the systemctl set-default multi-user doesn't work by itself.

lis6502, 2024-06-25:
not sure if still relevant, i've stumbled upon same issue and found workaround.
append to your ~/.bash_profile
`echo -e "\e[?16;0;110c"`
this sets software cursor to nonblinking solid cyan block.

lis6502, 2024-06-27:
and following the https://github.com/PotatoMania/uconsole-r01-bsp-kernel-for-archlinux

### Want a cursor on framebuffer

R-01 doesn't have a hardware cursor(that's default), and you can use a cmdline parameter to change the default
to a softcursor.

Format:: `vt.cur_default=0x{p3}{p2}{p1}`, note the order.

Example: `vt.cur_default=0xF00058` white background block, with highlight.

Cursor stays visible in and out of vi