clockworkpi/uConsole #7: hardware sources?

mid-kid, 2023-09-18:
I'm seeing the open source hardware logo all over the website, but I can only find the schematics. Are the PCB drawings and CAD software files available anywhere?

mid-kid, 2023-09-18:
Or the mechanical bits, for that matter. Where's the files required to make the different plates?

cuu, 2023-09-18:
not available

57Bravo, 2023-09-18:
I had the same question. I mean, having the a subset of the schematics is useful, but I wouldn't call that open hardware. 

The main reason I ordered a uConsole was that I believed these files would be pushed at some point. To date, I haven't seen much.

spv420, 2023-09-18:
> not available
will they be made available, or are we going to have to RE everything?

mid-kid, 2023-09-24:
I wanted to take a jab at fixing the noise in the headphone jack, relocate the wifi antenna plug, and order bare boards of the keyboard to try different switches, among other things that I wanna mess around with. Redoing the entire mainboard isn't exactly something I was looking forward to...

tablatronix, 2024-06-27:
Any updates to this? Screw sizes even ?

cuu, 2024-10-25:
https://github.com/clockworkpi/uConsole/blob/master/PCB/CPI_3.14_Mainboard_V5_Gerber.7z