clockworkpi/uConsole #9: Please add instructions on how to compile the keyboard firmware

SamusAranX, 2023-10-02:
See title.

cuu, 2023-10-03:
The development process is as same as devterm 

https://github.com/clockworkpi/DevTerm/wiki/Compile-keyboard-bootloader-and-firmware

## Difference
in later uconsole keyboard , I've edited bootloader with following code
```
diff --git a/config.h b/config.h
index b377f98..f8ce81a 100644
--- a/config.h
+++ b/config.h
@@ -156,7 +156,7 @@
 // Use Boot1 PB2 as the button, as hardly anyone uses this pin as GPIO
 // Need to set the button input mode to just CR_INPUT and not CR_INPUT_PU_PD because the external pullup on the jumplink is very weak
        #define BUTTON_INPUT_MODE       CR_INPUT
-    #define BUTTON_BANK GPIOB
+    #define BUTTON_BANK GPIOD
     #define BUTTON_PIN 2
     #define BUTTON_PRESSED_STATE 1

```

this modification can make cli flashing more stable, but cause one issue is I can not use Arduino IDE to flash the code

I need to run this command in a right timing manually

for example if you compiled the code in IDE ,try uploading once
upload will fail and
you will get  a console command output from Arduino IDE like (verbose output in Arduino IDE) in bottom: 

```
/root/.arduino15/packages/stm32duino/tools/stm32tools/2021.5.31/linux/maple_upload ttyACM0 2 1EAF:0003 /tmp/arduino_build_303517/devterm_keyboard_mini.ino.bin /home/cuu/data/Downloads/arduino-1.8.13
```

_I use root to run arduino ide btw_

so copy this command in terminal (dont close arduino IDE)

and plug the keyboard ,right after the LED on the keyboard started to flash

press Enter in terminal to run above cli command, then the flashing will success 

tricky way but it works

[uconsole_keyboard_flash.tar.gz](https://github.com/clockworkpi/uConsole/blob/master/Bin/uconsole_keyboard_flash.tar.gz) works
by this mod bootloader 

so try not to flash the bootloader unless you know what is going on

jsenecal, 2025-09-17:
I've tried the above instructions, and is seems like the binary generated by arduino isnt compatible anymore when replacing the file in [uconsole_keyboard_flash.tar.gz](https://github.com/clockworkpi/uConsole/blob/master/Bin/uconsole_keyboard_flash.tar.gz)
```
Failed to open serial device.
/usr/bin/dfu-util -d 1EAF:0003 -a 2 -D uconsole_keyboard.ino.bin -R
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: No DFU capable USB device available
Waiting for /dev/ttyACM0 serial...arm64
Done
```

This is arduino 1.8.20, and I loaded and selected the board from the above instructions.

An alternative would be to simply update the aforementioned tool included firmware with the new "Select" to scroll instead of the current "FN" to scroll.

Thanks!