clockworkpi/uConsole #11: No power / battery settings for CM4

scyto, 2023-10-11:
I expected to be able to set power and battery settings in the desktop on the CM4.

Was surprised when i couldn't.  Does the CM4 module support sleep states?

neggles, 2023-10-18:
CM4/Pi 4 have absolutely no support whatsoever for any kind of sleep state, no 😢

lambtor, 2024-01-26:
is there any way to configure the battery level icon by telling it what capacity your batteries claim to have?

neggles, 2024-02-07:
> is there any way to configure the battery level icon by telling it what capacity your batteries claim to have?

Yes, there is - but you have to modify the kernel/devicetree binaries to do it. I did this over in [my kernel fork](https://github.com/neggles/linux-rpi/commits/uconsole-6.6.y/) where I have a better version of the LCD driver and a few other things, too 😄