clockworkpi/uConsole #13: Running kodi screen display direction is incorrect

xue-fei, 2023-11-28:
CM4 version, running Kodi, screen display direction incorrect, How to adjust it?
System is [uConsole_CM4_v1.3g_64bit.img.7z](http://dl.clockworkpi.com/uConsole_CM4_v1.3g_64bit.img.7z)

57Bravo, 2023-11-28:
xrandr -o right

xue-fei, 2023-12-02:
> xrandr -o right

Have you tested it? I tried but it didn't work

andrwj, 2023-12-02:
`xrandr -o right` does work for most cases but it does not work for Kodi either on CPI image or vanilla RPi image.
I posted in the forums about this a long time ago, but no one had a good answer. 
My most recent conclusion is that the Kodi source for version 19 or so has a poor implementation of screen rotation. 
I've been meaning to try compiling version 20 Kodi at the smallest level, but it's a lot of work, so I've put it off until now.
I hope someone can point me to a solution.

57Bravo, 2023-12-02:
That's a bummer. Yeah, my image is just vanilla RPI with the CPI kernel/cfg patches. I guess you could look at any Kodi specific kernel patches and find mods made to /boot/config.txt made for their distro.

57Bravo, 2023-12-02:
Maybe this "display_rotate=2 in `/boot/config.txt"

Some folks reporting needing to use 4 instead of 2

https://raspberrypi.stackexchange.com/questions/136497/how-do-i-rotate-the-screen-at-boot-time-gui-is-rotated-but-boot-messages-are-st

andrwj, 2023-12-03:
@57Bravo Thanks for your replies. But `display_rotate=2` also does not work. 
Kodi seems to access the display device directly without using any kernel services or Windows services.