clockworkpi/uConsole #14: Where are PCBs?!

wacoinm, 2023-11-30:
you write open design, source
so where is board PCB?!

docteurzoidberg, 2024-02-17:
i would like to have at least a template for the extension modules....

DanieleQ97, 2024-02-17:
@docteurzoidberg I've made [this EasyEDA component](https://easyeda.com/component/65afb6bcfa9442839c95a99396b1e24b) to use as template for a project I'm working on.
It should be mostly right, but I haven't yet ordered any PCBs based on it.
I've only checked the dimensions using my inkjet printer, which isn't exactly the most precise tool 😅.

The pinout is based on the schematics for the 4G expansion board: [clockwork_UC_4G_Schematic.pdf](https://github.com/clockworkpi/uConsole/blob/master/clockwork_UC_4G_Schematic.pdf)

vitasam, 2024-08-01:
apparently not

shymega, 2024-09-02:
Duplicate of #7.

cuu, 2024-10-25:
https://github.com/clockworkpi/uConsole/blob/master/PCB/CPI_3.14_Mainboard_V5_Gerber.7z