clockworkpi/uConsole #17: Unable to connect to 4G network

otanim, 2024-02-03:
I followed the instructions here:
https://github.com/clockworkpi/uConsole/wiki/How-to-use-the-4G-extension

Managed to upgrade 4G extension firmware based on this article:
https://github.com/clockworkpi/uConsole/wiki/How-to-upgrade-4G-extension-firmware

Currently when I run `lsusb`, for the device I'm getting `9011` code instead of `9001`,
`ifconfig` shows that I now have `usb0`,

In the network manager's applet it even shows that I have an available network to connect (including ISP's name and the connection's strength), however, when I try to connect to it using the `nmcli c up <connection name>`, it throws this error:
`Error: Connection activation failed: IP configuration could not be reserved (no available address, timeout, etc.)`

Any Ideas?

cuu, 2024-02-04:
If you have 'usb0,' based on my experience, t
here is no need to use 'nmcli' to manually activate the 4G network. 
Simply turn off the Wi-Fi, and 'usb0' will be used automatically. 
Of course, make sure to insert the SIM card first.

otanim, 2024-02-07:
I've tried that, didn't work, apparently it doesn't work with some of SIM cards, I tried a different SIM card from a different network provider, it finally worked.

And I'm not 100% sure, but I think this device also sent an sms to some unknow number (+44749116344) which I definitely didn't have initiated, but it's been logged in my network provider's logs.

closed 2024-02-07 by otanim