clockworkpi/uConsole #27: Open-Source Hardware Compliance

DOSputin, 2024-10-16:
Dear clockworkpi.  Love what you are working towards, but there is much work to be done to keep up your end!  As a manufacturer advertising a product with the Open-Source Hardware (OSH) emblem, you are generally expected to provide design files, including PCB layouts, CAD sources, schematics, and other documentation necessary upon request (#7 #14 ) to allow anyone to study, modify, or replicate the hardware. The Open-Source Hardware Association (OSHWA) defines open-source hardware as hardware for which the design is made publicly available, enabling anyone to produce, modify, distribute, or learn from the hardware.

[image]

Please provide revision-controlled source files to comply with the principles of open-source hardware, ensuring that all relevant design files (PCB, CAD, schematics, etc.) are accessible, versioned, and properly licensed to allow for open collaboration and modification by the community. **Alternatively, if you are unwilling to provide these files, please do not advertise your products as Open-Source Hardware.**

cuu, 2024-10-25:
schematics ,PCB in here:

https://github.com/clockworkpi/uConsole

here is the 3D files:

https://www.clockworkpi.com/product-page/uconsole-3d-models-in-obj-format-free

what else do you need exactly?

DOSputin, 2024-10-25:
@cuu, your products are advertised with the Open-Source Hardware Association (OSHWA) emblem, prominently displayed about the website, etc.  It’s generally expected to provide comprehensive design files beyond the provided STEP files and PDF schematics. The OSHWA emblem indicates a commitment to the open-source hardware community standards, which require all source files needed to study, modify, or replicate the hardware.

Specifically, OSHWA-compliant “source files” include:
- **PCB Design Files**: The native files from the electronic design software used (such as KiCad, Altium, etc.) that allow modification, not just exported formats like STEP.
- **Schematics and CAD Files**: Editable CAD files and schematics that provide insight into the hardware design beyond static PDF exports.

For full transparency, it’s expected that these files be shared in a **public, revision-controlled** format, such as a repository on GitHub, to enable tracking of design changes, community collaboration, and modifications over time. This versioning is a key aspect of maintaining openness and aligns with community standards for traceability and usability.

If providing these source files isn’t feasible, I’d recommend reconsidering the OSHWA emblem to avoid confusion. Thank you for considering this alignment with OSH standards.  Because if you don't provide this stuff, using the OSHWA logo without full compliance might bring legal consequences, including potential trademark infringement (since OSHWA owns the rights to control how its logo is used), false advertising (misleading customers, like me, about the product’s open-source status), and even consumer fraud issues if many buyers feel misled. So, to avoid these risks, I suggest aligning fully with OSHWA’s guidelines or considering removing the logo from your product marketing asap.

cuu, 2024-10-25:
so kicad project file for uconsole ?

anything else?

DOSputin, 2024-10-25:
Providing revision-controlled source files for all EDA (Electronic Design Automation) designs, along with revision-controlled source files for all mechanical CAD designs, has been previously requested and is being explicitly requested by the community through this issue.  Thanks!

NyanHelsing, 2025-01-03:
hey is the source files for the keyboard in here too? i want do a mod to have 2 trackballs on mine

kondr1, 2025-05-01:
@cuu can you add keyboard source please?

neolao, 2025-07-31:
(I'm not a contributor of this project)