clockworkpi/uConsole #32: KiCAD files 4G modem

ArjenCNX, 2025-04-18:
Hi There Team Clockwork, just received your device, and i think its awesome, I would love to add to it's awesomeness by designing a Lora radio module for this device.   to do this it would be mighty helpful to have the KiCAD project files for the 4G modem, so i can edit them and turn it in to a Lora modem enabling long-range communication. 

Could you be so kind as to release the HW side of things there? i will of course share the design files for all once the work is complete. 

Thanks in Advance, 

Arjen

cuu, 2025-04-18:
Sorry but there is no kicad file for the 4G modem

ArjenCNX, 2025-04-21:
Do you have files in any other format?

cuu, 2025-04-22:
Sorry but right now only gerber files, no cad files avaiable.