clockworkpi/uConsole #35: 4G module

fanewang72, 2025-06-03:
Can someone to explain why you guys ship a version os this 4G extension modem that is old firmware and nothing is working as should work?
Why I need to update firmware to work with the device you sell? Also because that procedures of write the new firmware is pretty complicated and 98% chance to brick the 4G module! At least include in the original distribution a GUI app that upgrade this module, or you are just in search on customers that continue to complain about this module and probable 90% are forced to buy another one because they brick the old one upon your instructions that most of the time are not working even for people with years of knowledge in computers hardware and software.
I am based in China and as you know all the linux distort links are not working, so is already a pain to use any kind of linux distort, why I must to f--k my brain also to update a firmware at a board I bought few days ago from you?

cuu, 2025-06-04:
既然是在国内，你可以用中文

我大概理解了你的意思，你就不理解为啥要升级固件对吧？其实我们拿到高通这个4G模块也是无语的，我们也发现了固件版本不一致的问题，不能保证这个固件版本可以配合 linux modem manager gui 

所以最后求爷爷告奶奶的从高通那里搞了一个固件，9011还是多少我暂时忘了，可以看wiki

这个升级的wiki我本人还是测试很多遍，还是比较靠谱的，可以解决联网问题，直接插卡就能联网而不需要配置

如果因为操作难度等问题，我表示遗憾

有任何售后问题，可以直接 email 联系 help@clockworkpi.com

fanewang72, 2025-06-10:
I tried to follow that procedure but is not working, i arrive until here, that command doesn't work.

"First, send the command AT+BOOTLDR to /dev/ttyUSB3 to put the device into bootloader mode. At this point, the LED on the back will not light up.

echo -en  "AT+BOOTLDR\r\n" |sudo socat - /dev/ttyUSB3,crnl"