clockworkpi/uConsole #37: uconsole first startup, error updating initramfs, keyboard issues as a result

lolheyaj, 2025-08-05:
I just got my uconsole and on first startup the unit attempted to run updates and got an error when attempting to upgrade "initramfs-tools."

When moving on from this error and trying to install the keyboard flash utility, I get an error related to the an issue with "initramfm-tools" but I am unable to upgrade or update it, I just continue to get an error.

cuu, 2025-08-05:
which os image u are using ?

cuu, 2025-08-05:
if you just use the default os image come with factory SD card,then it is outdated, I recommend to flash a newer os image

such as 

https://forum.clockworkpi.com/t/bookworm-6-12-y-for-the-uconsole-and-devterm/15847

closed 2025-08-05 by lolheyaj